ricardoquesada/bluepad32-arduino #1: Support of XBox Elite 2 Controller

Erebos84, 2021-10-21:
Hello,
is the xbox elite 2 controller supported?
It is mentioned in the nina firmware but not in your list of supported controllers.
I'm able to connect it to the "Arduino Nano RP2040 Connect", but no inputs are recognized.

I used the precompiled nina firmware v2.2.1 with Bluepad32 library for Arduino v0.99.1

{ MAKE_CONTROLLER_ID( 0x045e, 0x0b05 ), CONTROLLER_TYPE_XBoxOneController, "Xbox One Elite 2 Controller" },	// { MAKE_CONTROLLER_ID( 0x045e, 0x0b13 ), CONTROLLER_TYPE_XBoxOneController, "Xbox One Series X Controller" },	// Microsoft X-Box One Elite Series X pad (Bluetooth)

ricardoquesada, 2021-10-21:
I don't have one to test it... the key question is: Is the Xbox One Elite Series 2 using Bluetooth (classic), or BLE?

If it uses Bluetooth (classic) then it should work.
If it uses BLE, then it won't work with v2.2.1... Adding BLE support is is my TODO list.

Erebos84, 2021-10-25:
First I tried it with an Arduino Nano RP2040. Now I have an ESP32 with the precompiled firmware v2.2.1. 
When I want to compile the gamepad example from Bluepad32 library I get several error messages. 
Is there a diffrent library for the ESP32?

*\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp: In static member function 'static void SpiDrv::begin()':
*\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:91:20: error: 'PINS_COUNT' was not declared in this scope
   if (SLAVERESET > PINS_COUNT) {

*\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp: 100:11: error: 'NINA_GPIO0' was not declared in this scope
   pinMode(NINA_GPIO0, OUTPUT);

*\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp: In static member function 'static int SpiDrv::available()':
*\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:66:22: error: 'NINA_GPIO0' was not declared in this scope
 #define NINA_GPIOIRQ NINA_GPIO0
 
*\Arduino\libraries\Bluepad32\src\utility\spi_drv.cpp:545:47: note: in expansion of macro 'NINA_GPIOIRQ'
 int SpiDrv::available() { return (digitalRead(NINA_GPIOIRQ) != LOW); }

Erebos84, 2021-11-06:
Hi,
i have now reflashed the precompiled bluepad32-unijoysticle (v2.2.1) on the ESP32. The Elite 2 controller does not pair with the ESP32. 
I have attached two logs to this post. The first one shows the pairing after the first boot of the ESP32. The second log shows the pairing attempt after a reboot of the ESP32. However, unfortunately I don't know enough about it to find the cause of the problem. Can you imagine what is the cause here?

(I have made a part of the MAC unrecognizable and shortened the first log a bit).

[first boot.log](https://github.com/ricardoquesada/bluepad32-arduino/files/7491438/first.boot.log)
[second boot.log](https://github.com/ricardoquesada/bluepad32-arduino/files/7491439/second.boot.log)

ricardoquesada, 2021-11-07:
I took a look them, and I seems that everything is "OK"... I mean, the Xbox One S has similar issues... my guess is that the Xbox One S and the Xbox Elite share the same firmware (or might be 99% identical).
I'd just say: try again... sorry, don't have a better answer.

Erebos84, 2021-11-07:
Does Bluepad32/btstack use Enhanced Retransmission Mode (ERTM)? In forums it is recommended to turn it off to be able to connect Xbox controllers with Linux.

ricardoquesada, 2021-11-07:
good find.
btstack supports ERTM, but it seems it is disabled by default... but I'll do some tests with it to double-check it.

ricardoquesada, 2023-03-15:
Although I don't have an Elite 2 to test, the new firmware (v3.6.0 and newer) supports BLE connections and with the Xbox firmware 5.x, it should work oK.

Please reopen it if it doesn't work. thanks

closed 2023-03-15 by ricardoquesada