ricardoquesada/bluepad32-arduino #2: Cannot connect to XBOX One S controller after firmware update

rodrigobomfim, 2021-11-22:
Hi, I've started with Bluepad32 few days ago and I was able to successfully use my XBOX One S controllers with Bluepad32, but pairing it was a hit & miss. I've then decided to update the firmware on my XBOX controllers and since then I cannot connect anymore.

Arduino Board: Nano RP2040 Connect
Firmware: Bluepad32 for NINA v2.3.0
XBOX Controller Firmware: Nov 2021(?)

Is this expected to work?

  Thanks

ricardoquesada, 2021-11-22:
I guess you are using firmware 4.9. And yes, it is "expected" that it fails more often than before.

But it is possible to connect it. Try connecting it to the Xbox and then try again with the bluepad32.... that might work.

ricardoquesada, 2023-03-15:
The new firmware (v3.6.0 and newer) supports BLE connections and with the Xbox firmware 5.x, it should work oK.

Please reopen it if it doesn't work. thanks

closed 2023-03-15 by ricardoquesada