ricardoquesada/bluepad32-arduino #3: Will this library work in the arduino ide  with an esp32 devmodule kit?

lexirobins, 2021-11-29:
Just curious if I can encorporate this into a project I'm working on.

ricardoquesada, 2021-11-29:
TL;DR: no

Long: This project only works for Arduino boards that have the Nina chip (a ESP32) as a co-processor.

But but but... I'm working on supporting Arduino + ESP32 modules...
At this moment we are discussing how to support Arduino in any ESP32 module (not just the official Arduino + Nina chips).

Feel free to join Discord to get the latest updates: https://discord.gg/r5aMn6Cw5q

lexirobins, 2021-11-30:
Ok. thanks for the swift reply! Can I take that to mean that the  ESP-IDF  instructions on the gitlab also won't work?

ricardoquesada, 2021-11-30:
Which instructions are you referring to? this one: https://gitlab.com/ricardoquesada/bluepad32/-/blob/main/docs/plat_nina.md ?
If so, they are only valid for "Nina" Arduino boards... that's not what you are looking for.

I'll announce it on the discord chat once it is ready for use... and also I'll close this bug.... so either feel free to join the discord chat, or keep an eye on this bug.

ricardoquesada, 2021-12-02:
Added experimental Arduino + Bluepad32 support. Please take a look at these instructions:
https://gitlab.com/ricardoquesada/bluepad32/-/blob/main/docs/plat_arduino.md

closed 2023-01-03 by ricardoquesada