ricardoquesada/bluepad32-arduino #5: Incorrect Bluetooth address when multiple devices connected

nimice, 2023-01-01:
**Prerequisites:**
- No device connected.

**Stimuli:**
- Connect one device with BT address X.
- Connect another device with BT address Y.

**Response:**
- When onConnectedGamepad is called for device X, the BT address for device X is used. **OK**
- When onConnectedGamepad is called for device Y, the BT address for device X is used. **NOK**

**Expected response:**
- The correct BT address is seen in onConnectedGamepad so that functionality can be built towards a specific device.

ricardoquesada, 2023-01-02:
which  bluepad32-arduino library version are you using? which bluepad32 FW version are you using?

nimice, 2023-01-02:
bluepad32-arduino library version is **1.1.4**
bluepad32 FW version is **3.5.2**

ricardoquesada, 2023-01-04:
Fixed in 1.1.5.

closed 2023-01-04 by ricardoquesada