ricardoquesada/bluepad32-arduino #7: Xbox Button not set properly

leep-frog, 2023-09-07:
When connected to a wireless Xbox controller (model 1708), the Xbox system button in the middle is not properly registered.

Debugging steps I performed:
- Verified all of the other buttons work and register as expected
- Verify the middle button works when connected to my Xbox
- Printed out the entire miscButtons bit mask and verified no bits were set when pressing the xbox button (but the select and start bits were properly set when pressing them)

ricardoquesada, 2023-09-08:
which platform are you using ?

- "Arduino IDE" with ESP32 ?
- "Arduino IDE" with NINA ?
- "Arduino + IDF" ?

and which version of Bluepad32 are you using ?

( See: https://tinyurl.com/bluepad32-help )

ricardoquesada, 2023-09-08:
I guess the most important question is which Xbox Firmware are you using

ricardoquesada, 2023-09-08:
Xbox button was working Ok.
But "select" button was not mapped. I've just fixed that. Will be included in v3.9.0

closed 2023-09-08 by ricardoquesada