ricardoquesada/bluepad32-arduino #8: Bluepad32 cannot disconnect() arbitrary controllers under Arduino environment.

paulpankratz, 2023-12-05:
When multiple controllers are connected to RP2040/NINA Nano using the disconnect() call it removes the first connected controller. Request is to alter the Arduino library to support disconnect(idx); where idx is the number of the controller from 0 to BP32_MAX_CONTROLLERS

ricardoquesada, 2023-12-06:
please attach the sketch that reproduces the bug

paulpankratz, 2023-12-06:
Ricardo, 

You will need to change the following to match your MAC PS4 controller address: 

byte AllowedAddress [ 6 ]= { 0x 88 ,0x 03 ,0x 4c ,0x 2d ,0x f2 ,0x 3b } ; 

commandparse () is where you can issue discrete commands. 

The example code that is included with the Arduino library exabits the same behavior. I haven't tried a different RP2040/NINI Nano. 

Thank You :) 

Paul

paulpankratz, 2023-12-06:
Sorry, I was responding to the automatically generated email thinking you would get it. I have attached it here :)
[SWAC_Wireless_Controller_PS4.zip](https://github.com/ricardoquesada/bluepad32-arduino/files/13591030/SWAC_Wireless_Controller_PS4.zip)

wiredopposite, 2024-01-21:
I have this issue with a generic esp32 dev board, here's the project 
[bluepad_i2c_master.zip](https://github.com/ricardoquesada/bluepad32-arduino/files/14000018/bluepad_i2c_master.zip)

ricardoquesada, 2024-01-22:
yes. it was always disconnecting gamepad 0. fixed in develop branch here:

https://github.com/ricardoquesada/bluepad32/commit/0075a5af9443b7c891d71dd6da02039bf90b8997

It will be part of v4.0... I'll release v4.0-beta0 in a few weeks.

closed 2024-01-22 by ricardoquesada

paulpankratz, 2024-01-22:
That's great news! I was going to reach out in the next couple of days to see how you were progressing :)