ricardoquesada/bluepad32-arduino #9: Compatibility with Arduino Uno R4 WiFi

Dario-Ciceri, 2023-12-22:
Hi, I would like to know if it is possible to make everything compatible with Arduino Uno R4 WiFi.
Thank you

ricardoquesada, 2023-12-23:
It should be possible, but it is not supported at the moment

Dario-Ciceri, 2023-12-24:
thanks, wanting to try to make it compatible, which file contains the services to check? I can locate the address of the controller, to reuse some of your code, what are the parts needed to connect with the controller and callback calls for the various advertisers?

ricardoquesada, 2023-12-24:
Take a look at how Bluepad32 for NINA works

https://github.com/ricardoquesada/bluepad32/blob/main/docs/plat_nina.md

I'm assuming that WiFi R4 uses something similar: SPI to talk to the ESP32-S3... if so, then it should be more or less straightforward...

Dario-Ciceri, 2023-12-24:
> Take a look at how Bluepad32 for NINA works
> 
> https://github.com/ricardoquesada/bluepad32/blob/main/docs/plat_nina.md
> 
> I'm assuming that WiFi R4 uses something similar: SPI to talk to the ESP32-S3... if so, then it should be more or less straightforward...

yes perfect, from what i know it uses a serial to communicate with module esp32 so i will try to adapt it, thanks. In case I will send you the code here

closed 2023-12-24 by Dario-Ciceri

ricardoquesada, 2023-12-24:
great. yes, feel free to send a pull-request. thanks.