ricardoquesada/bluepad32-arduino #10: Any plans for Arduino Uno R4 WiFi compatibility?

fstrugar, 2024-01-11:
R4 is now using Renesas RA4M1 main processor and ESP32-S3 co-processor. Thanks!

ricardoquesada, 2024-01-12:
if someone from the community would like to add support for it, happy to integrate the change.

Otherwise, low priority.

fstrugar, 2024-01-12:
Many thanks for the feedback! Any pointers/hints/ideas on what would be the best approach for implementing it?

ricardoquesada, 2024-01-13:
**a)**

would be good to know where the Uno r4 firmware lives.
E.g: the FW for the Arduino boards with NINA is here: https://github.com/arduino/nina-fw 

But where is the FW for the Uno R4 ESP32-S3 module ?

This is useful for different reasons:

- Pin used for SPI
- How it is being configured
- etc.

**b)**

Or the ARduino Uno R4 WiFi Schematic... from it you will be able to know which are the pins used for SPI.

**c)**

Once you know the SPI pins (perhaps are the same as NINA) make the needed changes to:

https://github.com/ricardoquesada/bluepad32/blob/develop/src/components/bluepad32/platform/uni_platform_nina.c

If the SPI pins are the same, perhaps you don't have to change anything...

use `develop` branch... don't use `main`

**d)**

and that's it (?)

ricardoquesada, 2024-01-13:
mmm... might be more difficult than expected.. it might not be using SPI, but USB to connect both MCU.

IIUC, this is the source code:
https://github.com/arduino/uno-r4-wifi-usb-bridge

and the NINA platform is not suitable for that... so a new platform should be created, and that talks to the host using the same USB protocol (this is to simplify things).

fstrugar, 2024-01-14:
Thank you for all the info - it's a bit beyond my current understanding but if I manage to get enough time to play with it at least I know where (not) to look.

I've played a bit with ArduinoBLE to try to get some understanding of what's happening and it seems to be using HCI (https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/utility/HCI.h) for host controller communication - if I understand things correctly.

I've also tried using ArduinoBLE to connect to an xbox controller (BLE) just to see what happens, and ArduinoBLE sees it, talks to it, but ends up freezing in infinite loop in [discoverDescriptors](https://github.com/arduino-libraries/ArduinoBLE/blob/dff9e414abc20df491b46d0b38b360eb1a3507fb/src/utility/ATT.cpp#L1721)... it seems to be a known issue.

ricardoquesada, 2024-01-15:
if you want to use Bluepad32, then you should not use ArduinoBLE... they are not compatibles.

fstrugar, 2024-01-15:
Yeah I was just looking at how they handle the communication between RA4M1 and ESP32-S3.

I've also just found https://github.com/h2zero/NimBLE-Arduino and then realized it doesn't support Uno R4 either.

So maybe the most logical thing for me would be to just return the Uno R4 and get one with a NINA chip :)