ricardoquesada/bluepad32-arduino #11: Xbox button is not registered

microbmen, 2024-01-18:
I have an xbox controller model 1708,  ive tried a few of them.  
when pressing the xbox center button there is nothing registered under miscButtons.  When I try a PS4 or PS5 the PS button DOES register.  It only seems to be an xbox 1708.  

Im running bluepad33-arduino NINA

further, 

How do you call the xbox button direct? or is it on under misc()?  
ctrl->miscButtons() or can there be a ctrl->xbox() or ctrl->systembutton()?

ricardoquesada, 2024-01-19:
which bluepad32 version are you using ? I think it was fixed in one of the latest version. Use Bluepad32-nina v3.10.3 (in case you are not using it)

microbmen, 2024-01-19:
ahh, im using the newest version of bluepad32-arduino v 1.31.  I guess not the same amount of love in that repo.  Ill check out moving over to the main branch.

ricardoquesada, 2024-01-19:
the client is "bluepad32-arduino", but you need to flash the firmware in the NINA chip.

https://github.com/ricardoquesada/bluepad32/blob/main/docs/plat_nina.md

microbmen, 2024-01-19:
Im on 3.10 firmware, but still the xbox doesnt work..

configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
BTstack up and running at FC:B4:67:F5:4C:9E
2D AE 65 3B FB EF 74 1E 22 7D 48 AF f� 19 90 
**Firmware: Bluepad32 for Arduino v3.10.2**
BD Addr: FC:B4:67:F5:4C:9E
2D AE 65 3B FB EF 74 1E 22 7D 48 AF A1 9C 19 90

microbmen, 2024-01-19:
hang on , i see you said 3.10.3 ... let me see whats up

microbmen, 2024-01-19:
im at 3.10.3 and still no xbox button.  all other buttons seem to register fine.   I wanted to queue off the xbox button to do a disconnect.

Bluepad32 (C) 2016-2023 Ricardo Quesada and contributors.
Version: v3.10.3
BTStack: Copyright (C) 2017 BlueKitchen GmbH.
Platform: unijoysticle2
Unijoysticle: Board ID values: 1,1,1
Hardware detected: Unijoysticle2

microbmen, 2024-01-19:
figured it out. 

when xbox (model 1708) controller is on firmware 3.1.1221 the xbox button doesnt register.

when xbox controller is on version 5.17.3202 the xbox button DOES register.  

interesting...

ricardoquesada, 2024-01-19:
mmm... weird. I wonder why... probably a bug from Bluepad32 (?).

you said that you are using NINA... which NINA board exactly are you using ?

Because if you are using NINA, then two things:
  - You flashed the "unijoysticle" FW... and not the NINA one (from: https://github.com/ricardoquesada/bluepad32-arduino/issues/11#issuecomment-1899738939 )
  - You should not be able to see those logs from NINA

microbmen, 2024-01-19:
im building from src, and customizing using `idf.py menuconfig`, setting the max of connected gamepads to 2, then `idf.py build`, then` idf.py -p COM03 flash`  ...

microbmen, 2024-01-19:
sorry updated above, [HiLetgo-ESP-WROOM-32-Bluetooth-ESP32-DevKitC-32-Development](https://www.amazon.com/HiLetgo-ESP-WROOM-32-Bluetooth-ESP32-DevKitC-32-Development/dp/B0CNYK7WT2?pd_rd_w=1vH66&content-id=amzn1.sym.32628caa-c59e-4a45-93eb-93ac6202e29a&pf_rd_p=32628caa-c59e-4a45-93eb-93ac6202e29a&pf_rd_r=YKTRHX8F3ATK9KKG16PH&pd_rd_wg=UaMgC&pd_rd_r=0735ea3f-cc40-4e1a-8e80-caafd2b9db11&psc=1&ref_=pd_bap_d_csi_pd_ys_c_rfy_rp_crs_0_t)

not sure the exact NINA inside to those specifics.

ricardoquesada, 2024-01-19:
ok. You are **NOT** using NINA. You are using a regular ESP32 board.

The firmware that you flashed is almost Ok. You flashed a firmware that is suitable for Unijosyticle boards.... not good for regular ESP32 boards.

Assuming that you want to use Arduino, then you should flash this firmware (includes Bluepad32, Arduino core,  and other dependencies):

- https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template

Assuming that you are NOT interested in Arduino, then you should use `develop` branch, and use this example as reference:
- https://github.com/ricardoquesada/bluepad32/tree/develop/examples/esp32

In any case you should be able to see the "Xbox" button pressed with any of those  two repos.

microbmen, 2024-01-19:
ok, will try this firmware.  i have another board that is Nano 33 (no wifi), i think that's where i got that from,  but for this project im using this ESP32 board.  

trying this other esp32 firmware gives errors,
```
   Arduino-esp32 can be used with ESP-IDF versions between 4.4.0 and 4.4.99,
    but a newer version is detected: 5.1.2.

```

ill see if i can get it to compile

still - weird that only the xbox button is the one that fails to work...

ricardoquesada, 2024-01-19:
yes, `https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template` requires ESP-IDF v4.4 for the moment.

This is because the Arduino Core (v2.x) only works with ESP-IDF v4.4.

Espressif is working on the new Arduino Core v3.x, and that one will support ESP-IDF 5.x.. so yes, use ESP-IDF 4.4.6

microbmen, 2024-01-19:
ok running on the Arduino v3.10.2 firmware 

configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
BTstack up and running at FC:B4:67:F5:4C:9E
5F 5E 52 CB A9 77 88 2E 21 D1 1C 05 18 �� F8 
Firmware: Bluepad32 for Arduino v3.10.2
BD Addr: FC:B4:67:F5:4C:9E
5F 5E 52 CB A9 77 88 2E 21 D1 1C 05 18 2A 1C F8 

Firmware info:
        Bluepad32 Version: v3.10.2 (3.10.2)
        Arduino Core Version: v2.0.14
        Compile Time: Jan 19 2024 13:52:25

IDF Version:v4.4.6
Chip info:
        model:ESP32
        cores:2
        feature:/802.11bgn/BLE/BT/External-Flash:4 MB
        revision number:3

xbox button still doesnt work on xbox controller firmware -  3.1.1221

microbmen, 2024-01-19:
Do you have an xbox 1708 controller?  if so, downgrade the firmware so it supports classic bluetooth and try it?

xboxaccessories://firmwareupdate/?legacyDowngrade=true

ricardoquesada, 2024-01-19:
yes, I can try.

In any case, why are you using FW 3.1 ? You can try FW 4.8 (which is still BT classic), and or 5.17 (BLE)... all of them work better than FW 3.1

microbmen, 2024-01-19:
3.1 and 5.17 are the only 2 supported by MS xbox controller utility.  I don't even know how your installing 4.8.

microbmen, 2024-01-19:
i want the controller to also support older iot devices which are only classic bt (no BLE).  without the need to constantly flash the firmware.

ricardoquesada, 2024-01-19:
which IOT device are BT classic only ?

microbmen, 2024-01-19:
older arduino devices that ive built - i could go upgrade everything to BLE but thats another can of worms.

ricardoquesada, 2024-01-19:
be more specific... because Bluepad32 doesn't run in "old IoT" devices... it only runs on ESP32 / PicoW devices... all of them support Classic and BLE... in fact some of them only support BLE.

microbmen, 2024-01-19:
I understand.  You asked why I still want to run classic BT on the Xbox controller.   I'm not talking about bluepad32.  im talking about the xbox controller supporting classic BT such that the controller can still connect to older stacks like an atmega328 running UHS.

ricardoquesada, 2024-01-19:
got it. I'll try to test it on FW 3.1... but you should know that FW 3.1 is kind of buggy... not a good controller ... the axis are not linear... if you find the way to upgrade to v4.8, I strongly recommend it.... avoid v3.1 whenever possible

microbmen, 2024-01-19:
I've not seen issues, been using it for years.   How do you upgrade to 4.8?  is there some utility to flashes xbox controller firmware? I use the Xbox accessories app for windows.

microbmen, 2024-02-05:
FYI.  Xbox firmware 3.1 system() button still an issue in 4.0 beta.

Works fine in 5.1.

ricardoquesada, 2024-02-05:
@microbmen yep. I didn't have the time to  test it.

In any case, perhaps you can fix it yourself. 
the bug should be on this function:
https://github.com/ricardoquesada/bluepad32/blob/main/src/components/bluepad32/parser/uni_hid_parser_xboxone.c#L128

If you are using the Arduino IDE... well... it will be difficult to compile a new core... but you can try it from here:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template